open-ephys/GUI #330: ADC channel noise

nikolaskaralis, 2016-02-17:
It seems that analog-in channels are very noisy, but only when acquisition is on.
Specifically, I tried the following:
1. Connect an oscilloscope using a BNC cable to one BNC port of the IO board attached to the analog-in HDMI port.
2. The signal in the oscilloscope is low noise.
3. Start acquisition on OpenEphys GUI.
4. This leads to very noisy traces in the oscilloscope (and the same is recorded by the GUI).

I have tested this issue with 3 different boards and it seems to be general.
Different grounding combinations didn't resolve this issue.

Any ideas?

aacuevas, 2016-02-17:
Shouldn't this issue be on the board tracker instead of the software one?

Anyway, the analog inputs, being inputs, don't have any specified value when left floating, so it's no surprise they are just noisy in that case. The difference between acquisition active or not is probably due the slight variations on the ADCs inputs due the sensing switching. If they were noisy when an input is plugged, that would be worrysome, however. Grounding the unused ADC inputs (just connecting them to the IO board ground) should make the noise disappear. Does that happen in your case?

As for a solution for floating inputs noise, there is none except grounding the channels or disabling them via channel mapper on the GUI. We are planning, though, to add some isolation and protection to all inputs in a future revision of the hardware, so we could think of a way to leave them in a default 0V state when unplugged.

nikolaskaralis, 2016-02-17:
Hi Aaron,
thanks for the response.
I am not sure I get why they are noisy when acquisition is active.
I didn't try grounding them (will do tomorrow), however when they are
connected to a device they are still noisy (that is how I spotted the issue
in the first place).
Importantly this noise is leaking to the oscilloscope (where it is visible
when acquisition is active but not when inactive), so this is not just
noise recorded by the amplifiers, but rather noise that originates in the
board.

As for the choice of board, perhaps it is a hardware issue, but the fact
that it is depended on the acquisition state made me think it is a software
issue as well.

aacuevas, 2016-02-17:
Ok, if they are noisy when connected to a signal too, then there is an issue.
Have you tried disabling the LEDs? There were cases when those were inducing quite an amount of noise on the ADCs, that's why we provided the option to disable them.

nikolaskaralis, 2016-02-17:
Yes, disabling the LEDs doesn't seem to make any difference.

aacuevas, 2016-02-17:
Which kind of signal was the device generating and what is the output impedance of that device? Please try with the inputs grounded to a baseline of 0V and see if the GUI reads any noise. Then try again with any other fixed voltage in the range -5V,+5V and measure if there is noise both in the GUI and with an oscilloscope. I'll run some tests of my own meanwhile.

As for why there is noise when acquisition is active and not when it isn't, that's easy. When acquisition is inactive, most of the board just sits idle with a fixed voltage in almost all traces and most of the ICs just waiting in a sleep state. When acquisition starts, so does the signal party: clock and digital signals are continously sent to the ADCs, DACs, LEDs and headstages. Those devices, in turn, start working, generating their own share of internal signals. All of those are noise sources of different magnitudes. Any input pin is electrically connected to its source via low impedance at one side and to the circuitry (and thus the noise sources) via a very high impedance. If there is signal in the input, that difference makes so that only the signal is actually present in the line but if the input were to be left floating (which equals infinite impedance) then the very high impedance that isolates the input suddenly becomes comparatively low (the other is infinite!) so that the noise can flow from the circuit into the pin.

jvoigts, 2016-02-17:
The oscilloscope noise sorta makes sense to me - you're essentially connecting it to the aq. board on a high impedance pin, so it's bound to pick up a lot of noise.

The npise when you ground an input shouldnt happen - if the inputs are grounded then the reading should be flat, pretty close to 0.
What are the input jumpers set to?  This can change the behaviour of the system between straight to the DAC, via unity gain opamp, or via a voltage divider that remaps +-5 to 0-5V. Maybe there's somehitng going on related to this? Here is a diagram of the relevant Ain section: 
[image]

jonnew, 2016-02-17:
yes, Aaron is right - you need to terminate the BNC into a low impedance.
This is why you need to amplifiy neural signals before recording them (its
just as much for impendance conversion as it is to bring the signal into
the dyanmic range of your ADC...). Does your oscope provide a 50ohm
termination feature? If so, switch it on and watch the noise go away.

That said:

I dont have the layout in front of me, but the onboard DACs and ADCs share
a ground plane with FPGA, don't they? Without some form high-freq isolation
between the analog and digital ground/power (e.g. split planes connected
with ferrite chips or back to back schottkys), it will be very hard to get
the full 16 bits of resolution out of those ADCs and its not suprising to
me that digital noise is getting onto the

http://www.ti.com/lit/an/slyt499/slyt499.pdf

Mixed signal design at that resolution is not easy. Very careful ground and
power plane layout is required.

linussun, 2016-05-12:
Just putting in my two cents: I was having a extremely noisy ADC input for a voltage/eye position signal during acquisition, which was not seen as much when I would test the eye position signal on an oscilloscope. It turns out that a low quality power supply for a Dell LCD monitor that was plugged into a wall power socket far away (not even on the same rack that was supplying power to the eye coil/Open Ephys Intan system) was the culprit. By unplugging that power supply, the signal/noise dramatically improved on the ADC inputs.

rishidhingra, 2018-03-22:
I've been experiencing a similar issue and wondering if anyone has any insight. 

In my case, when I connect a probe thru an intan headstage and also use an IO board to acquire data from a pair of differential amps, the data from the intan headstage becomes contaminated with noise consisting of high frequency square pulses. If I disconnect the hdmi cable, the system is perfectly ground and yields clean signals....

jvoigts, 2018-05-30:
That's a good observation! - makes sense that the open ended clk signal would bleed over especially in less well shielded cables.
You should be able to turn off the i2c in the firmware pretty easily, but making that a configurable register that you can set from the software would require a little bit of work, or maybe we should just kill the i2c clock out by default for now  (@aacuevas what do you think?) - if someone wanted to use it they would already have to edit the firmware anyways.

Instead of cutting the trace you could also desolder the i2c level converter - its the small ic under the fpga plug in board, this way its easier to reverse later.

aacuevas, 2018-05-30:
I've checked the firmware and we don't even have an I2C controller in it. That line was added for future expansion but right now the pin is just outputting a copy of the headstages' SPI clock for debug purposes.
I'll disable it and upload new bitfiles to the development branch.

rishidhingra, 2018-05-31:
Does this mean I can potentially disable the relay of the clock signal by updating the FPGA firmware? If not, I'll try @jvoigts suggestion to desolder the i2c IC. Thanks!!

jvoigts, 2018-05-31:
Awesome, thx Aaron! Yes this means that you can fix the issue by just replacing the bitfile, assuming this i2c clk signal is also the source of your noise, but it seems likely. I assume @aacuevas can just reply here when the bitfile is up.

aacuevas, 2018-05-31:
I've updated the bitfiles, disabling any output to those pins, they should be set as high impedance now.

rishidhingra, 2018-06-08:
Sorry it took me a bit to get back to the rig. The updated bitfiles work like a charm on the dev branch. I just needed to load the visual studio 2015 redistributable to get the compiled version to load without errors. Thanks for the help!!

closed 2022-07-30 by jsiegle