open-ephys/GUI #331: Add safeguard for very high spike detection rates

jvoigts, 2016-02-23:
Running the spike detector/viewer/sorter with a very low threshold, or running it with intermittent high noise levels can generate very high rates of spike events. This can lead to cpu load spikes that then can lead to data loss.

To avoid this issue we could implement different safeguards:

- Add a setting with a simple X Hz sustained rate over Y msec threshold that stops further spikes from being detected as long as the rate is exceeded.

- Somewhat related, it would be nice to implement an automatic spike threshold selection that chooses the threshold based on some fairly simple quantile metric - i think Quiroga had a very simple and robust method for that in some paper?

shayo, 2016-02-23:
Quiroga proposed to use some fraction of the median (
http://www.scholarpedia.org/article/Spike_sorting#Step_ii.29_Spike_Detection
)

An efficient implementation would be to use a running median per channel.
Something like this: http://playground.arduino.cc/Main/RunningMedian

kylerbrown, 2016-02-23:
+1 Automatic spike thresholding is a great idea.

jonnew, 2016-02-23:
NeuroRighter did this. We found that having a simple exponentially weighted moving average on the RMS noise for each channel with a ~10 second time constant worked really well. The user could then select the threshold has some multiple of the RMS noise. Transient bursting events did not affect the threshold, but sustained events (like seizures) would.

jsiegle, 2022-07-30:
Automatic threshold setting (based on signal standard deviation or the Quiroga method) is now available in the Spike Detector.

closed 2022-07-30 by jsiegle