open-ephys/GUI #333: LFPViewer: Crash when zooming in an ADC channel

malfatti, 2016-02-26:
If I add FPGA source and LFP viewer, then start acquisition, change the voltage range to >1V and double-click an ADC channel , the GUI freezes and seconds after I got several lines of:

`Error in Rhd2000EvalBoard::readDataBlock: Incorrect header.`

and I have to kill the interface. If I do not double-click a channel, everything works wonderfully.

I'm using Debian, kernel 3.14.17-xenomai-2.6.4, OpenEphys board v2.2. I just cloned and compiled the GUI, no errors in the output.

Thanks :)

jvoigts, 2016-02-26:
Just found the same bug yesterday but assumed that it was due to something i had done in my branch. I'll look into it - seems like the method for resetting the channels can sometimes report a 0px channel height which then causes other issues.

jvoigts, 2016-02-27:
solved the bug I had, https://github.com/jvoigts/plugin-GUI/commit/fae727b83352ae782626af6a312cb031f3a5670a
Don't think its related though, my issue wasn't restricted to ADC channels.

jvoigts, 2016-02-27:
I just pushed the fix to the main GUI branch in ef8174f879914517d0c773ae0adf1d6356113906 - see if you can still get the error or if this is fixed now.

malfatti, 2016-02-28:
Indeed, your commit fixed the bug and improved A LOT scrolling the LFP window :D 

Still, one not-so-important important detail: If I doucle-click an ADC channel and change the voltage range, the gui freezes forever, without crash and with no error message in terminal. If I set the desired voltage first, and then double-click, it works perfectly. 
The strange thing is that with headstage channels I can double-click, change voltage range, change spread, timebase, anything and it works super fine, the problem is specific to ADC channels :/

Thanks a lot for fixing this!!

jvoigts, 2016-02-28:
alright, i think i can reproduce this, but not reliably. I can get the main branch to crash _sometimes_ if i just set the AUX range to the smallest value and then scroll down to make the AUX channels visible. This means its probably some out of bounds drawing issue that doesn't manage to raise an exception. 

As a quick test, can you try to replicate the bug in the newer lfp drawing branch ( https://github.com/jvoigts/plugin-GUI/tree/improved-lfp-rendering ) where I added a bunch of out of bounds checks? If this solves the problem we know it has to be in the channel rendering code and it will be an easy fix.

malfatti, 2016-02-29:
I tried, now, but it seems to have only FPGA and file reader as sources and Merger and Splitter as utilities, no filters and no sinks. I also get a suspicious "Plugin path not found" and a "File not found" in the console. 

I tried `git clone -b improved-lfp-rendering https://github.com/jvoigts/plugin-GUI/`, and downloading the zip directly from the branch, and got the same results. Am I doing something wrong? Should I merge this branch with a fork of GUI:master?

malfatti, 2016-02-29:
I just tested, it works perfectly. I could zoom in, change everything, zoom back, did it several times and it is perfect!

Thanks a lot!! :D

jvoigts, 2016-02-29:
Ah yea, the plugin branch requires some specific compiling. In case someone else runs into this, here's the documentation on how to compile the plugins: https://open-ephys.atlassian.net/wiki/display/OEW/Plugin+architecture

So it sounds like the updated lfp viewer solves the problem - i'll try to add simple range checks into the main branch then.

Please note that the improved lfp branch is still pretty unstable, so maybe be extra careful if you're using it for recordings now. We'll eventually fold the new method into the main branch but it might take a while to make sure its stable. If you do run into specific problems with the new lfp viewer, please report bugs on the branch: https://github.com/jvoigts/plugin-GUI/issues

closed 2022-07-30 by jsiegle