open-ephys/GUI #341: Network Events source on Mac OSX

mikewehr, 2016-07-06:
Any advice on how to get the Network Events source to show up on a Mac OSX install of the master binary? I have installed zmq using macports, and also tried with zmq3, but Network source does not appear in the GUI. It appears all the right files are in /opt/local/lib, and matlab_zeroMQ_wrapper_example.m executes fine, which suggests that zmq is working ... what else can I check about the zmq installation?
Works fine for me on Ubuntu, was hoping for platform independence. 
-Mike

$ ls  /opt/local/lib
libsodium.18.dylib	libzmq.3.dylib		pkgconfig
libsodium.a		libzmq.a
libsodium.dylib		libzmq.dylib

jonnew, 2016-07-06:
Are you using the plugin-GUI? I think this is a plugin that will need to be
compiled separately from the main GUI.

Jon

mikewehr, 2016-07-06:
Thanks Jon. I don’t understand why it would need to be compiled separately if I’m using the downloaded binaries, and the Network Events source works fine with an ubuntu install. That’s why I’m suspicious of the zmq dependencies on Mac, but I’m at a loss how to troubleshoot those.
-Mike

mikewehr, 2016-07-07:
Indeed, it works after compiling from source. Thanks Jon!

jonnew, 2016-07-07:
@mikewehr: yeah, its always the best solution. Maintaining all those binaries is a full time job that I'm glad I have nothing to do with :) @jsiegle and @jvoigts You may want to reopen this since the pre-compiled binaries don't seem to come with the plugins which takes away about 80% of their functionality.

mikewehr, 2016-07-07:
Can't people still find the thread if it's closed?
    
    
    On 7/6/16 8:31 PM, Jon wrote:
    
    
      @mikewehr: yeah, its always the best
        solution. Maintaining all those binaries is a full time job that
        I'm glad I have nothing to do with :) @jsiegle
        and @jvoigts You may want to reopen this
        since the pre-compiled binaries don't seem to come with the
        plugins which takes away about 80% of their functionality.
      —
          it on GitHub, or mute
          the thread.

jonnew, 2016-07-07:
1. Not really because it gets buried in the "closed" section which is not
searched by default on github.
2. Understand that issues are not "bad" they are just a statement of fact.
This is not peer review :).

mikewehr, 2016-07-07:
I'm re-opening this issue for awareness. The problem is solved by compiling from source code.

jsiegle, 2016-07-07:
Hmmm...which OS X version are you using? It might be due to the fact that I compiled zmq for 10.11. I've been meaning to re-compile to make it backwards-compatible. Is there a chance this is the issue?

mikewehr, 2016-07-07:
Well, recompiling solved the problem for me (El Capitan 10.11.5). So my best guess is that was the issue.

closed 2022-07-30 by jsiegle