open-ephys/GUI #342: Crash on recording: Windows 7

pickledenthusiasm, 2016-08-09:
Hi,

The current binary (ver 0.4.1.0) seems to crash every time I try to record. Every once in a blue moon it records a few seconds of data first, but mostly it crashes after writing the settings.xml file. I've repeated the problem on a few different Windows 7 SP1 machines, and I don't think it's a hardware problem because we're able to record using the same hardware on a linux machine.

The error message blames MSVCR120.dll but I'm no sure that's the real cause. I've tried removing that file and allowing it to use a copy (which appears identical) downloaded with Visual C++ Redistributable, but that doesn't help.

Any ideas would be much appreciated!

Thanks,
Luke

Faulting application name: open-ephys.exe, version: 0.4.1.0, time stamp: 0x575a0bbb
Faulting module name: MSVCR120.dll, version: 12.0.21005.1, time stamp: 0x524f83ff
Exception code: 0x40000015
Fault offset: 0x0000000000074a46
Faulting process id: 0x21c8
Faulting application start time: 0x01d1f1da1f827e7a
Faulting application path: C:\Program Files\Open Ephys\open-ephys.exe
Faulting module path: C:\Windows\system32\MSVCR120.dll
Report Id: 70b8c1cf-5dcd-11e6-a2e0-0090f5c3ef25

jsiegle, 2016-08-09:
Does it crash when using both data formats, or just one? And how many channels are you trying to record? Windows has problems opening 100s of files at once.

Also, what are the details of your processor? The latest version of the GUI uses a separate thread for recording, which has led to issues on CPUs with fewer cores.

pickledenthusiasm, 2016-08-09:
Thanks for the quick response! I've isolated the problem to the network events source. 

The processor is a quad core, Intel i7-2700K. It crashes using either data format, even with just one channel recorded at the lowest sampling rate.

It doesn't crash if the network events source is removed, and crashes even if the network events module is the only item in the graph. Are there older versions of that module that might work? I will also try building it myself from the source.

Also, we're considering running the GUI in Linux instead. Are there any pros or cons we should be aware of before going that route?

Thanks!

shayo, 2016-08-09:
Make sure network events isn't set to record any channels. This is a known
issue.

Shay

pickledenthusiasm, 2016-08-10:
That did the trick, thank you.

closed 2022-07-30 by jsiegle