open-ephys/GUI #345: Mac GUI crashes after clicking 'cancel' when no acquisition board detected

atheliarosapaulli, 2016-08-30:
Hi Open Ephys team!
If I want to do a clean install of Open Ephys, are there hidden configurations stored somewhere that I need to get rid of? I have run into an issue where the GUI (running on a Mac) fails to detect an acquisition board, and then crashes when I select the cancel option in the pop-up stating no acquisition board has been found. The issue is reproducible, and I have successfully gotten the GUI to detect my acquisition board on another Mac, so I believe there is a bug in the configurations I have saved on the Mac where it continually crashes. I have attached the error report at the bottom of the post, for your convenience. 
Thank you!

**Error Message**
Process:               open-ephys [1264]
Path:                  /Applications/osx-el-capitan-master/open-ephys.app/Contents/MacOS/open-ephys
Identifier:            org.open-ephys.gui
Version:               0.4.1 (0.4.1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           open-ephys [1264]
User ID:               501

Date/Time:             2016-08-29 15:03:38.386 -0400
OS Version:            Mac OS X 10.11.6 (15G31)
Report Version:        11
Anonymous UUID:        BC357FAD-DF72-C6EA-2BDA-C00F9C86141F

Sleep/Wake UUID:       6880A0BB-FC94-4901-91F5-83F2D32AA98D

Time Awake Since Boot: 10000 seconds
Time Since Wake:       2300 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Juce Message Thread  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0:
--> 
    __TEXT                 00000001005db000-0000000100eab000 [ 9024K] r-x/rwx SM=COW  /Applications/osx-el-capitan-master/open-ephys.app/Contents/MacOS/open-ephys

Thread 0 Crashed:: Juce Message Thread  Dispatch queue: com.apple.main-thread
0   org.open-ephys.gui            	0x0000000100616a64 okCFrontPanel::SetWireInValue(int, unsigned int, unsigned int) + 20
1   org.open-ephys.gui            	0x000000010062337f Rhd2000EvalBoard::setClockDivider(int) + 31
2   org.open-ephys.gui            	0x00000001006105d4 RHD2000Thread::setClockDivider(int) + 84
3   org.open-ephys.gui            	0x0000000100603d27 RHD2000Editor::loadCustomParameters(juce::XmlElement*) + 1063
4   org.open-ephys.gui            	0x0000000100632ad8 GenericEditor::loadEditorParameters(juce::XmlElement*) + 152
5   org.open-ephys.gui            	0x0000000100640011 GenericProcessor::loadFromXml() + 305
6   org.open-ephys.gui            	0x000000010064c933 ProcessorGraph::restoreParameters() + 179
7   org.open-ephys.gui            	0x000000010068646a EditorViewport::loadState(juce::File) + 4618
8   org.open-ephys.gui            	0x0000000100695620 MainWindow::MainWindow() + 1312
9   org.open-ephys.gui            	0x0000000100696daa OpenEphysApplication::initialise(juce::String const&) + 314
10  org.open-ephys.gui            	0x000000010083c3a6 juce::JUCEApplicationBase::initialiseApp() + 118
11  org.open-ephys.gui            	0x00000001009cb28b juce::JUCEApplication::initialiseApp() + 11
12  org.open-ephys.gui            	0x000000010083c08f juce::JUCEApplicationBase::main() + 127
13  org.open-ephys.gui            	0x000000010083bfc3 juce::JUCEApplicationBase::main(int, char const**) + 83
14  libdyld.dylib                 	0x00007fff8a3855ad start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff95c945e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff95db0578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff95dae341 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff95c94efa kevent_qos + 10
1   libdispatch.dylib             	0x00007fff95cd8165 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff95cd7dcd _dispatch_mgr_thread + 52

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff95c945e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff95db0578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff95dae341 start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff95c945e2 __workq_kernreturn + 10
[... 374 lines cut ...]
===========                     =======  ======= 
TOTAL                            691.9M      649 

Model: MacBookPro9,1, BootROM MBP91.00D3.B0D, 4 processors, Intel Core i7, 2.6 GHz, 8 GB, SMC 2.1f175
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Graphics: NVIDIA GeForce GT 650M, NVIDIA GeForce GT 650M, PCIe, 1024 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333531533643465238432D50422020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54333531533643465238432D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xF5), Broadcom BCM43xx 1.0 (7.21.95.175.1a6)
Bluetooth: Version 4.4.6f1 17910, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Samsung SSD 850 EVO 250GB, 250.06 GB
Serial ATA Device: OPTIARC DVD RW AD-5970H
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: IR Receiver
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 3.0 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc., 25.1

jsiegle, 2016-08-31:
If you right-click on the application and select "Show package contents," there's a file called windowState.xml in the MacOS folder that determines whether or not a previous configuration is loaded. You could try deleting that and see if it fixes the problem. There aren't any configuration files stored outside of the Open Ephys application.

Are you using a USB 2.0 or 3.0 acquisition board?

atheliarosapaulli, 2016-09-02:
We are using a USB 2.0 acquisition board.

I tried your suggestion, but there was no file called windowState.xml in the MacOS folder. Please let me know if you have any other advice. 

Thank you!

atheliarosapaulli, 2016-09-27:
Hi! We're not sure exactly what fixed it, but our issue has been resolved.

closed 2022-07-30 by jsiegle