open-ephys/GUI #346: Couldn't load a file from file reader on xubuntu 16.04

delcasso, 2016-11-02:
I cannot load a file using ifle reader.
The window to select the file doesn't open on xubuntu.
Please could you help me ?
Thank you.
Sebastien

Here a copy what appear on the terminal when I used open ephys.

./open-ephys 
JUCE v3.0.6

Created processor graph.

Got audio device.

Audio device name: Playback/recording through the PulseAudio sound server
Audio device type: ALSA
Audio output channels: 2
Audio device sample rate: 44100
Audio device buffer size: 1024

Created audio component.
Created message center.
Created info label.
Created graph viewer.
Adding tab with index 1
Adding tab with index 2
Created data viewport.
Created filter viewport.
Created control panel.
Created filter list.
Created plugin manager
Loading Plugin: FilterNode... Loaded with 1 plugins
Loading Plugin: PulsePalOutput... Loaded with 1 plugins
Loading Plugin: PhaseDetector... Loaded with 1 plugins
Loading Plugin: KWIKFormat... ../../Source/Processors/PluginManager/PluginManager.cpp:177: Failed to load plugin DLL: libhdf5.so.10: cannot open shared object file: No such file or directory
 DLL Load FAILED
Loading Plugin: ChannelMappingNode... Loaded with 1 plugins
Loading Plugin: LfpDisplayNode... Loaded with 1 plugins
Loading Plugin: Rectifier... Loaded with 1 plugins
Loading Plugin: SpikeSorter... Loaded with 1 plugins
Loading Plugin: RecordControl... Loaded with 1 plugins
Loading Plugin: ArduinoOutput... Loaded with 1 plugins
Loading Plugin: EventBroadcaster... Loaded with 1 plugins
Loading Plugin: NetworkEvents... Loaded with 1 plugins
Loading Plugin: SerialInput... Loaded with 1 plugins
Loading Plugin: CAR... Loaded with 1 plugins
Loading Plugin: LfpDisplayNodeBeta... Loaded with 1 plugins
Loading Plugin: BasicSpikeDisplay... Loaded with 2 plugins

Loading window bounds.

Item dropped at insertion point 0
Creating from description...
Sources::File Reader (0-3)
  Adding node to graph with ID number 100

File Reader updating settings.
File Reader setting num outputs to 16
^Cdelcasso@delcasso-MacPro:~/programming/plugin-GUI/Builds/Linux/build$ ./open-ephys 
JUCE v3.0.6

Created processor graph.

Got audio device.

Audio device name: Playback/recording through the PulseAudio sound server
Audio device type: ALSA
Audio output channels: 2
Audio device sample rate: 44100
Audio device buffer size: 1024

Created audio component.
Created message center.
Created info label.
Created graph viewer.
Adding tab with index 1
Adding tab with index 2
Created data viewport.
Created filter viewport.
Created control panel.
Created filter list.
Created plugin manager
Loading Plugin: FilterNode... Loaded with 1 plugins
Loading Plugin: PulsePalOutput... Loaded with 1 plugins
Loading Plugin: PhaseDetector... Loaded with 1 plugins
Loading Plugin: KWIKFormat... ../../Source/Processors/PluginManager/PluginManager.cpp:177: Failed to load plugin DLL: libhdf5.so.10: cannot open shared object file: No such file or directory
 DLL Load FAILED
Loading Plugin: ChannelMappingNode... Loaded with 1 plugins
Loading Plugin: LfpDisplayNode... Loaded with 1 plugins
Loading Plugin: Rectifier... Loaded with 1 plugins
Loading Plugin: SpikeSorter... Loaded with 1 plugins
Loading Plugin: RecordControl... Loaded with 1 plugins
Loading Plugin: ArduinoOutput... Loaded with 1 plugins
Loading Plugin: EventBroadcaster... Loaded with 1 plugins
Loading Plugin: NetworkEvents... Loaded with 1 plugins
Loading Plugin: SerialInput... Loaded with 1 plugins
Loading Plugin: CAR... Loaded with 1 plugins
Loading Plugin: LfpDisplayNodeBeta... Loaded with 1 plugins
Loading Plugin: BasicSpikeDisplay... Loaded with 2 plugins

Loading window bounds.

Item dropped at insertion point 0
Creating from description...
Sources::File Reader (0-3)
  Adding node to graph with ID number 100

File Reader updating settings.
File Reader setting num outputs to 16

jsiegle, 2016-11-02:
Can you try it with the latest version of the "development" branch? That's based on Juce 4, which may fix the window issue. The latest File Reader also supports loading of files that are dragged and dropped onto the editor.

That said, it looks like it's failing to load the hdf5 library, so the KWIK format may not work anyway. Do you have the hdf5 library installed on that machine?

delcasso, 2016-11-02:
Sorry Josh, I don't see the development branch.
Are we talking about master branch, last version ?

Yes I have hdf5-1.8.17 on my machine and success installation checks.

jsiegle, 2016-11-02:
The code is available here: https://github.com/open-ephys/plugin-GUI/tree/development

You can obtain it locally by running:
```
git fetch origin development:development
git checkout development
```

delcasso, 2016-11-02:
I am compiling it.

I had this issue : 
In file included from ../../JuceLibraryCode/juce_gui_basics.cpp:9:0:
../../JuceLibraryCode/modules/juce_gui_basics/juce_gui_basics.cpp:104:37: fatal error: X11/extensions/Xrandr.h: No such file or directory
   #include <X11/extensions/Xrandr.h>

I sloved it with : sudo apt-get install libxrandr-dev

I am compiling plugins now

delcasso, 2016-11-02:
I am still having the same problem in loading the file.

Here is the log from terminal

./open-ephys.so 

_JUCE v4.2.1

Created processor graph.

Got audio device.

Audio device name: Playback/recording through the PulseAudio sound server
Audio device type: ALSA
Audio output channels: 2
Audio device sample rate: 44100
Audio device buffer size: 1024

Created audio component.
Created message center.
Created info label.
Created graph viewer.
Adding tab with index 1
Adding tab with index 2
Created data viewport.
Created filter viewport.
Created control panel.
Created filter list.
Created plugin manager
Loading Plugin: FilterNode... Loaded with 1 plugins
Loading Plugin: PhaseDetector... Loaded with 1 plugins
Loading Plugin: KWIKFormat... ../../Source/Processors/PluginManager/PluginManager.cpp:177: Failed to load plugin DLL: libhdf5.so.10: cannot open shared object file: No such file or directory
 DLL Load FAILED
Loading Plugin: Rectifier... Loaded with 1 plugins
Loading Plugin: SerialInput... Loaded with 1 plugins
Loading Plugin: CAR... Loaded with 1 plugins
Loading Plugin: BasicSpikeDisplay... Loaded with 2 plugins

Loading window bounds.

File not found.
Item dropped at insertion point 0
Creating from description...
Sources::Rhythm FPGA (0-0)
/home/delcasso/programming/plugin-GUI-development/Builds/Linux/build
---- Intan Technologies ---- Rhythm RHD2000 Controller v1.41 ----

FrontPanel DLL loaded.  Built: Feb  1 2012  17:34:05

Scanning USB for Opal Kelly devices...

Found 0 Opal Kelly devices connected:

Device could not be opened.  Is one connected?
  Adding node to graph with ID number 100

Rhythm FPGA updating settings.
Rhythm FPGA setting num outputs to 0
Setting buffer address to 0x5c8bff0
Input buffer address is 0x5c8bff0
Removing processor with ID 100
RHD2000 interface destroyed.
Item dropped at insertion point 0
Creating from description...
Sources::File Reader (0-3)
  Adding node to graph with ID number 101

File Reader updating settings.
File Reader setting num outputs to 16_

jsiegle, 2016-11-02:
What happens if you try to drag and drop the file onto the File Reader editor?

delcasso, 2016-11-02:
nothing more in  the terminal or in the gui
I probably need to solve my PluginManager.cpp error.
Loading Plugin: KWIKFormat... ../../Source/Processors/PluginManager/PluginManager.cpp:177: Failed to load plugin DLL: libhdf5.so.10: cannot open shared object file: No such file or directory

jsiegle, 2016-11-02:
It seems like it could be an hdf5 versioning issue. Do you have any libhdf5.so files, and are they in your path? Check `/usr/local/lib` or `/usr/local/hdf5`

delcasso, 2016-11-02:
in /usr/local/hdf5/lib
I have 
lrwxrwxrwx 1 root root      17 Nov  1 12:42 libhdf5.so -> libhdf5.so.10.2.0*

delcasso, 2016-11-02:
I also have libhdf5.so.10 -> libhdf5.so.10.2.0*

delcasso, 2016-11-02:
Sorry for being stupid
I dind't do :

export CPLUS_INCLUDE_PATH=/usr/local/hdf5/include
export LIBRARY_PATH=/usr/local/hdf5/lib
export LD_LIBRARY_PATH=/usr/local/hdf5/lib

before compiling plugins

delcasso, 2016-11-02:
I still cannot load a file with the file reader 

I am trying to drag and drop 
plugin-GUI-development/Resources/DataFiles/data_stream_16ch_hippocampus.kwd

or click "F: button"

here is my log

JUCE v4.2.1

Created processor graph.

Got audio device.

Audio device name: Playback/recording through the PulseAudio sound server
Audio device type: ALSA
Audio output channels: 2
Audio device sample rate: 44100
Audio device buffer size: 1024

Created audio component.
Created message center.
Created info label.
Created graph viewer.
Adding tab with index 1
Adding tab with index 2
Created data viewport.
Created filter viewport.
Created control panel.
Created filter list.
Created plugin manager
Loading Plugin: FilterNode... Loaded with 1 plugins
Loading Plugin: PhaseDetector... Loaded with 1 plugins
Loading Plugin: Rectifier... Loaded with 1 plugins
Loading Plugin: SerialInput... Loaded with 1 plugins
Loading Plugin: BasicSpikeDisplay... Loaded with 2 plugins

Loading window bounds.

File not found.
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
Item dropped at insertion point 0
Creating from description...
Sources::File Reader (0-3)
  Adding node to graph with ID number 100

File Reader updating settings.
File Reader setting num outputs to 16
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
JUCE Assertion failure in juce_GraphicsContext.cpp:36
Updating connections:

Signal chain: 0

Source node: File Reader.
     Connecting to audio and record nodes.
     No dest node.

Enabling processors...
File Reader said it's not OK.
Control panel received signal to disable callbacks.

delcasso, 2016-11-02:
Maybe that can help...
When I run the program I have see in the terminal 

Loading window bounds.
File not found.

delcasso, 2016-11-02:
in Source/MainWindow.cpp, line 185.
The program can't load windowState.xml

delcasso, 2016-11-02:
I run the program again and I don't have this error anymore, that's good, maybe the file didn't exist the first time.

Sorry for confusing you.

delcasso, 2016-11-02:
It's comming.
I saw people commenting about this on 
https://forum.juce.com/t/cant-get-jucer-to-open-files-on-linux/13534/5

It's probably a xubuntu problem.
I will try their suggestions

delcasso, 2016-11-02:
I modified the juce_FileChooser.h
But I don't know hoe to recompile the Introjucer now :)

delcasso, 2016-11-02:
I am recomplining open-ephys now

delcasso, 2016-11-02:
problem solved !!

delcasso, 2016-11-02:
I modified the file 
plugin-GUI-development/JuceLibraryCode/modules/juce_gui_basics/filebrowser/juce_FileChooser.h

`    FileChooser (const String& dialogBoxTitle,
                 const File& initialFileOrDirectory = File::nonexistent,
                 const String& filePatternsAllowed = String::empty,
		//2016/11/02 Sebastien Delcasso update according
	        //https://forum.juce.com/t/cant-get-jucer-to-open-files-on-linux/13534/6
	        //This
	        bool useOSNativeDialogBox
                #ifdef JUCE_LINUX
                    =false,
	         #else
	               =true,
	          #endif	
	          //Replace this
		   //bool useOSNativeDialogBox = true,							 
                   bool treatFilePackagesAsDirectories = false);`

closed 2016-11-02 by delcasso