open-ephys/GUI #353: Opal Kelly library file not found on Ubuntu 16.04 64-bit

agonzal2, 2018-03-07:
Hi there, I've been trying to install the binaries for my 64-bit machine with Ubuntu 16.04. I've followed the procedure on the website but the GUI can never find the library despite unplugging, rebooting, etc... I've also tried to overwrite rhd2000_usb3.bit and libokFrontPanel.so from Github to the executable folder. Neither has worked... Is it necessary to build from source? Any ideas to get at this? Thanks so much!
Alfredo

tomchallands, 2018-03-15:
I had the same problem on the same system. It is because you need to install libudev0. Though not explicitly stated that you have to install it on the Rythm FPGA page, it is stated that USB3 for Open Kelly requires libudev0.

Download from:

 https://ubuntu.pkgs.org/12.04/ubuntu-updates-main-amd64/libudev0_175-0ubuntu9.10_amd64.deb.html

Install using:

sudo dpkg -i libudev0_175-0ubuntu9.10_amd64.deb 

Then the OpenEphys GUI will connect to your acquisition board.

Tom

agonzal2, 2018-03-16:
That worked! Thanks so much Tom!

closed 2018-03-16 by agonzal2