open-ephys/GUI #354: Saving data every X minute

Hanlin-Zhu, 2018-05-14:
Hi Open-ephys developers,

       Is there any easy way to save data automatically every X minute as we are doing long term continuous recording with your system. Currently the data is only saved when manually starting and stopping the recording. As chronic continous recording becoming more and more prevalent, this would be a great feature to make the system more robust to unexpected shut down and also good for data transfer/management. Thanks! 

Best,
Spark

aacuevas, 2018-05-31:
Which file format are you using?
Data is actually saved continuously but some formats need some header updates that only occur at specific times.

aacuevas, 2018-06-01:
The open ephys format (.continuous files) writes to disk continuously for each buffer and does not require any header updates. Data saved in that format has been proven to be available even before a crash, so I find it weird that in your case you only see the data when you manually stop recording.

Which OS are you using?
If you tried to forcibly close the software while recording (via windows task manager or linux kill command), what would the continuous files contain?

Hanlin-Zhu, 2018-06-01:
Sorry. I mean for program to automatically save to a new file every X minutes. Since at this point we have to manually restart the program in order to save to a new file. We saw such feature in Intan and love to have in Open-ephys as well. Thanks. 
[image]

jsiegle, 2022-07-30:
This can either be achieved using the Record Control plugin, or by starting/stopping recording via the built-in HTTP server.

closed 2022-07-30 by jsiegle