open-ephys/GUI #355: Opal Kelly missing library

yannunh, 2018-05-31:
I am trying to use the pre-compiled GUI on Windows 10 and I get the Opal Kelly missing library window.
I already read this : https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/491621/Windows . It wasn't helpful. Should I download else? I downloaded open-ephys-windows.zip in http://www.open-ephys.org/gui, and run FrontPanelUSB-DriverOnly-4.4.0, then run open-ephys. 
[image]

aacuevas, 2018-05-31:
Umm, the file is definitely there, the okFrontPanel.dll
Can you please click Yes on that dialog, annotate the folder that opens by default and then try to manually open that dll?
If that does not work, please open the GUI with the --console command argument (create a shortcut, open its properties and add --console to the end of the target line) and send me (aaron@open-ephys.org) the full content of the console log as well as the default path that opens when trying to manually browse for the dll.

yannunh, 2018-06-01:
Thank you. I solved problem.
The problem was dir name, mine was written in Korean.
I saw console log and realized.  :D

Thank you.

aacuevas, 2018-06-01:
Glad this was solved.
It is good to know that some parts of the GUI can have trouble with non-ascii path names, we weren't aware of that.
Thanks!

closed 2018-06-01 by aacuevas