open-ephys/GUI #360: Organization/Reading .dat files from Flat Binary Readouts

pratikmistry08, 2018-10-26:
Hello,

I'm trying to visualize some recently collected data, and I'm running into some issues. When I tried to open the continuous.dat file in Neuroscope with the settings.xml file, I get the following results.
[image]

Which doesn't look correct. I was wondering if there are any edits I can make to the XML file to visualize it better.

Or what the strategy for parsing the continuous.dat file to extract the same waveforms we see from the LFP viewer

aacuevas, 2018-10-29:
From the looks of your file, it seems like an [endianness](https://en.wikipedia.org/wiki/Endianness) issue. Basically, the order the different 8-bit halves of the 16-bit data are stored. Our GUI stores the data in little endian format, which is the standard data format used by most PC applications, it seems that Neuroscope, as well as other tools, use big endian for historical reasons.

While it could be a good idea to add an option to select the endianess, you should be able to use matlab to convert the endianness with a code like this:
`src=memmapfile('source_filename','Format','int16');`
`dest=fopen('dest_filename','wb');`
`fwrite(dest, src.Data, 'int16', 0, 'b');`
`fclose(dest);`
`clear src;`

pratikmistry08, 2018-10-29:
Thanks for the response! That didn't work, but it turned out that there were 3 auxiliary channels that are on the shanks that were not being read into Neuroscope.

aacuevas, 2018-10-30:
Yes, You can select which channels to record by clicking the vertical lines at the right of the FPGA processor, which opens the channel selector. On the "REC" tab, uncheck any channels you don't want recorded.
Incidentally, you can use this same method to record channels from other processors, like filtered data after a bandpass filter.

EspeHope, 2022-02-22:
Anyone know how can I convert .dat files from Neuroscope to something readable on MATLAB?

closed 2022-07-30 by jsiegle