open-ephys/GUI #364: Can't use Intan Rec. Control in Windows 10

KotsosWir3, 2019-03-19:
Hi, 

I tried to use the open Ephys Gui to visualize LFP signals and check for spikes online in Windows 10 but I can't recognize the RHD2000 intan board. I read in an old issue that it could be due to a conflict of the drivers. Did anybody experience such a thing? 

Cheers! 
[image]

jsiegle, 2019-03-19:
Try the "Rhythm FPGA" module instead. The Intan Rec. Control module is meant for this: http://intantech.com/recording_controller.html

KotsosWir3, 2019-03-20:
Thanks a lot jsiegle! I already tried also the "Rythm FPGA"  module but with no success. Should I first add a file reader>spliter>LFP viewer or should I add Rythm FPGA>Spliter> LFP Viewer? 
Thanks a lot in advance!

aacuevas, 2019-03-20:
The File reader is for reading examples or prerecorded files. For using the RHD2000 evaluation board you need the Rhythm FPGA node.

With the included drivers (fontpanel v4.5.5) the board should suffice. Does the board appear in Windows' device manager when you plug it? Also, please run the GUI with the --console command line argument (by creating a shortcut to the exe and adding it at the end of the target line) and copy its contents, it could have hints on what is happening.

closed 2022-07-30 by jsiegle