open-ephys/GUI #365: Corrupt .continuous ADC and AUX data files of doubled size

cleppla, 2019-04-05:
I know that other threads have mentioned timestamp issues related to corrupt data files and that a fix for those has been created, but unless I've overlooked something I believe I may be experiencing a different issue as the fix script does not resolve the problem. 

The ADC and AUX channels that are flagged as corrupt are exactly double the size of the other files recorded. The file size doubling is occurring for ADC1-8 and AUX1-3, but not AUX4-6 or CH1-64 (recording with 2 32ch headstages). 

The issue only occurs occasionally (maybe 1 in 50-100 recordings), but when it does it will occur in all subsequent recordings until I catch it, closing and reopening the GUI does not resolve it, and it occurs in both short (3-5min) and long (1-1.5hr) recordings. I have not noticed the CPU load spiking, and the problem appears to go away if I rescan for headstages. Given the intermittent nature of the problem I have had difficulty pinpointing a cause. 

Has anyone else experienced this, and if so, can the data be saved? Any help would be greatly appreciated!

nicoleprovenza, 2020-02-24:
@cleppla were you able to find a fix for this issue? I am experiencing a similar issue.

cleppla, 2020-02-26:
Hey @nicoleprovenza , If you are experiencing the same file size doubling behavior that I described, I am fairly certain it is caused by unplugging and plugging in headstages without rescanning while the GUI is open. I was never able to recover any of the corrupt data though, so unfortunately can't provide any insights there. Hope that helps!

closed 2022-07-30 by jsiegle