open-ephys/GUI #368: TTL Events not showing up in Continuous Data

mattvinacco, 2019-07-02:
Hi All,

I am trying to record VEP's using an external trigger and I cannot find a way for the triggered events to show up in my continuous data.  Within the GUI, I am using the FPGA module merged together with the Serial Port module.  I am then trying to use the Event Triggered Avg module along with a spike detector, ending with the LPF Viewer.  The stimulus is being run on a MatLab program and (hopefully) the triggers would be sent into the recorded data to be later analyzed.  I have the trigger connected to the TTL port (HDMI closest to the headstages) using an Open Ephys I/O Board.  The LPF viewer shows the highlighted channels showing that it is receiving TTL events but after I record the data, I am not Able to see at which point in time the trigger was received.  I am using the Matlab plugin EEGLAB to convert the '.continuous' files into '.mat' files to read the continuous data.  That is where I am not able to see the event info, only the continuous data.  Is this possible to accomplish using the parameters I've set? 

Thanks,
Matt

reinderdorman, 2019-07-25:
I think you want to look in the all_channels.event file (or something similar). I use OpenEphys analysis tools, which has matlab functions to load this (and converting to .mat is only a save function).

Theres some good info here: https://open-ephys.atlassian.net/wiki/spaces/OEW/pages/65667092/Open+Ephys+format

jsiegle, 2022-07-30:
Most of these plugins have been substantially updated since this issue was posted. If you're still experiencing this problem with the latest version of the GUI, please open a new issue in the [`plugin-GUI`](https://github.com/open-ephys/plugin-GUI) repo.

closed 2022-07-30 by jsiegle