open-ephys/GUI #369: LFP viewer ADC channels not displayed to scale

kimpossible11, 2019-07-09:
Hello, 
I'm recording an analog signal off an ADC channels and I'm getting a weird scaling in the LFP viewer. Basically, if I put in a 500mV signal AC signal from a signal generator (default -5V/5V I/O board, and -5V/5V setting in FPGA setttings), it appears as a 4V in the LFP viewer (seen in the attached image at 2V range in the LFP viewer). However, both in the recorded file (OE file type) as well as on an oscillosope, the correct 500mV appears. The matlab plot y axis is in Volts. Is this a quirk of the viewer or am I missing something?

[image]
[image]

chentao-CQU, 2019-08-13:
Hello, I am also using open ephys this device, I am now experiencing a problem, when I build a signal chain and start collecting signals, 8 ADC channels start to display rectangular pulses, no matter if I connect the I/O  board ,it's always like this, which means that the analog signal is not input to the acquisition board at all. I want to know how to get the acquisition board to receive the analog input and display it on the GUI as shown in your figure.

jsiegle, 2022-07-30:
If you're still experiencing this problem in the latest version of the GUI, please open a new issue in the [`plugin-GUI`](https://github.com/open-ephys/plugin-GUI) repo.

closed 2022-07-30 by jsiegle