open-ephys/GUI #372: Gui does not connect to PulsePal v2.0

MobsLab, 2019-11-18:
Hello,

We were trying to connect GUI v.0.4.5 to our PulsePal2.0 using PulsePal plugin and failed. The stimulator does not react to any commands from the plugin.

Unfortunately, the software does not spit out any errors or incompatibilities, so we have no idea where to start our debugging from.

We know that Arduino Due board is a part of Pulse Pal. So we checked whether the software recognizes Arduino, and yes, we were able to successfully connect Due through 'Arduino 'Output' plugin.

The issue is reproduced both on Windows and Linux machines.

Do you have an idea what could cause that?

Thank you very much for your help,
Dima Bryzgalov

aacuevas, 2019-11-19:
Hello,

Could you please open this issue in the [plugin-GUI](https://github.com/open-ephys/plugin-GUI) repository? This one is for a very old version and unused.

In the meanwhile, please open the GUI with a debug console and send me its contents, and there might be clues on your issue.

MobsLab, 2019-12-05:
Ok, thank you!

closed 2019-12-05 by MobsLab