open-ephys/GUI #373: Board Detection and LFP Signal Issues on Acquisition Board

jesse-gilmer, 2019-11-25:
Hello,
I have been having issues connecting the GUI to the acquisition board, and well as in detecting LFP signals.
I will usually get this error message:
[image]
but this can be temporarily resolved by cycling the board power via unplugging, at which point the board will connect, but the LFP will quickly grow until it becomes candy cane-like: 
[image]
with the following error:
[image]

I have recreated the board detection error on windows 7 pro and ubuntu 16.04. I have downloaded VS2012 and 2013, but that did not resolve the issue. The usb port does not make a difference. The board will show purple LEDs post-failure on the IO ports, the intan ports seem normal. 
Any help would be appreciated.
Best,
Jesse Gilmer

aacuevas, 2019-11-25:
This seems like a hardware issue. Does the opal kelly appear on Windows' device manager?
Is it an usb2 or usb3 board?

jesse-gilmer, 2019-11-25:
@aacuevas Initially it doesn't appear on the device manager (and also doesn't light up), but after being loaded into the GUI it appears as "FX3 Bootloader." This is a USB3 board.

aacuevas, 2019-11-26:
Can you please send me to aaron (at) open-ephys.org the full console contents on a time when the device is detected by the GUI?
Could you also check if at that time the board appears in the device manager under FrontPAnel devices as XEM6310 ?

It might be a hardware issue, but I would like to confirm my suspicions.

closed 2022-07-30 by jsiegle