open-ephys/GUI #376: Signal scaling issue / HPF issue

ntotah, 2020-12-14:
Hi, I have two issues to report. I wonder if anyone else has had this problem. 

First issue is signal scaling:
I have put a 10Hz sine wave 130mV into the Intan RHD2132 via a low resistance (<0.3 Ohm) cable and recorded the signal on Channel 12 (.continuous) recorded in OpenEphys format. I used the most recent GUI on a MacBook Pro. I have imported the signal into MATLAB using the code load_open_ephys_data_faster.m in MATLAB 2019a. In the attached plot, you can see that the signal (plotted in uV) is 100x too small. I also show the signal on an oscilloscope before entering the headstage. What is also strange is that I should be able to see a sine wave of, for example, 500 uV in the GUI but I have to push my function generator up to >50 mV to see anything in the GUI window even when I set the range in the GUI to 500 uV.

Second issue is that clicking the DAQ HPF to say 50Hz does not remove the 10Hz oscillation from the function generator.

It would be great to know if anyone encountered these issues before and solved them.

Thanks!
Nelson

[image]
[image]

ntotah, 2020-12-16:
Hi jon,
You hit the nail on the head. I added a 1MOhm resistor. The signal looks great now. so that topic can be closed. I appreciate your help on it. 

The DAQ HPF setting to 100 Hz still allows the 10Hz sine from the function generator. This doesn't matter since we will record broadband, but it is curious that it isn't filtering the signal.

Nelson

jonnew, 2020-12-16:
By DAQ, what do you mean?

ntotah, 2020-12-17:
Hi Jon, I meant DAC HPF in the Rhythm FPGA. 
[image]

closed 2022-07-30 by jsiegle