open-ephys/GUI #378: Hatching error

dsmorrie, 2020-12-29:
Hello, We have a problem opening open-ephys.
To be precise, it is data acquisition problem.
We get a screen like this, we guess it cannot get appropriate data from each channel.

[image]

Is there anyone who knows the solution, or has experienced the same issue?
Thanks a lot for your help.

jvoigts, 2020-12-29:
The hatching itself just indicates samples that are on the maximum or minimum of the amplifier input range, so you are clipping and losing data there. This can happen if you have a lot of noise and/or you have non-connected channels. If you are seeing this despite the recording setup looking correct otherwise, I'd say try debugging the analog end, grounding etc. first. Possibly try grounding individual channels and verify that they go to zero correctly.

closed 2020-12-29 by jvoigts

dsmorrie, 2020-12-31:
Thank you so much! We tried this at another computer and it worked. We would also try grounding individual channels. We really appreciate your help.