open-ephys/GUI #383: v0.5.3 crashing while launching

shirquinn, 2021-01-12:
Hi
Iv'e used the new version once and it worked fine, ever since, when i launch the gui it crashes instantly when trying to load parameters for rhythm pfga. what can be causing this?
thank you

jtmiles, 2021-01-19:
I'm having the same issue as @shirquinn. Had a couple successful recording sessions but the GUI started crashing after trying to add an ADC channel with signals from a camera. Not sure where I can access the Rhythm FPGA parameters to take a look at them, either. Any advice would be great!

jtmiles, 2021-01-20:
Ah, just found this comment to a similar sounding issue (open-ephys/plugin-GUI#402) in the updated repository

shirquinn, 2021-01-20:
Thank you @jtmiles !

closed 2021-01-20 by shirquinn