open-ephys/GUI #385: Event syncronization

ananmoran, 2021-02-08:
Hi,
I am using Neuropixels and have different streams recorded, both the hipass AP and the LFP, as well as digital signals using a NI acquisition board, all boards reside in the same NI box. I configured the synchronization in the recording node, and all squares are green, signaling correct synchronization. Following recording I can see synchronized_timestamps.npy in all continuous channels, but not in the events/TTL channels. Is there a reason for that? What is the best way to perform event synchronization with the continuous signals, and consequently spike times?

Thanks
Anan

jsiegle, 2021-02-09:
The TTL event timestamps refer to the same sample indices as `timestamps.npy` for the continuous data. So finding the relevant synchronized timestamp just involves two steps:

1) Subtract the very first timestamp, so the timestamps are shifted relative to the recording start index
2) Use the result to index into `synchronized_timestamps.npy`

In Python, this would be something like:

```python
event_ts = np.load('/path/to/event/timestamps.npy')
continuous_ts = np.load('/path/to/continuous/timestamps.npy')
synchronized_ts = np.load('/path/to/continuous/synchronized_timestamps.npy')

synchronized_event_ts = synchronized_ts[event_ts - continuous_ts[0]]
```

ananmoran, 2021-02-10:
Hi Josh,
Thanks for your answer and code. Unfortunately using continuous_ts[0], the first sample of the continuous timestamps crashes with "out of bounds" error. However, I used the 0 time in the sync_messages.txt file to normalize the timestamps which gave better results. Does this seems a logical solution?

Thanks
Anan

jsiegle, 2021-02-10:
Are `synchronized_ts` and `continuous_ts` from the NIDAQ sub-processor or the Neuropixels sub-processor? If the TTLs came through the NIDAQ card, both of those need to come from the NIDAQ as well.

ananmoran, 2021-02-13:
They are from the NIDAQ. I have tested again following your instructions and it works perfectly!
Thanks Josh

closed 2022-07-30 by jsiegle