open-ephys/GUI #387: open ephys crash

ntotah, 2021-05-02:
Hi all, we left the software running for a 1.5 hours and came back to find it frozen (i.e., the LFP traces were not being updated. We were able to import the files, but they were only 18 min long. We had been in acquisition mode for about 3 hours, then hit record and rat a recording for about 45 min, then hit stop, changed the file name, and hit record again for the 1.5 hr recording. During that recording, the software crashed. Is there a way for me to share the settings xml file here?

jsiegle, 2021-05-02:
The settings file should have been written when the recorded started—you can just upload that as an attachment to this issue.

ntotah, 2021-05-03:
Hi josh,I didn’t see a way to attach the file to the issue report. I’m attaching to this email. Please let me know that you can see it. Thanks,Nelson  From: Josh Siegle ***@***.***>Date: Sunday, 2. May 2021 at 21.40To: open-ephys/GUI ***@***.***>Cc: ntotah ***@***.***>, Author ***@***.***>Subject: Re: [open-ephys/GUI] open ephys crash (#387)The settings file should have been written when the recorded started—you can just upload that as an attachment to this issue.

jsiegle, 2021-05-03:
It doesn't look like the attachment came through.

You should be able to drag and drop it into the text field you're editing on GitHub. But if that doesn't work, you can just email it to support@open-ephys.org

jsiegle, 2021-05-03:
The signal chain seems fine. Was there any other software running concurrently on this computer? And did you happen to notice anything that was being output to the console when the crash happened?

I did notice the operating system is Windows 8, which we haven't done any testing with. Is there a reason this computer hasn't been upgraded to Windows 10?

ntotah, 2021-05-04:
Hi Josh,It is actually windows 10. We only had safari running. (well, and all of the extras required by the IT department…see screen shot of startup apps and screen shot of running processes) . I also put a screen shot of the windows details  From: Josh Siegle ***@***.***>Date: Monday, 3. May 2021 at 21.50To: open-ephys/GUI ***@***.***>Cc: ntotah ***@***.***>, Author ***@***.***>Subject: Re: [open-ephys/GUI] open ephys crash (#387)The signal chain seems fine. Was there any other software running concurrently on this computer? And did you happen to notice anything that was being output to the console when the crash happened?I did notice the operating system is Windows 8, which we haven't done any testing with. Is there a reason this computer hasn't been upgraded to Windows 10?

jsiegle, 2021-05-12:
Ok, it looks like the OS version is always saved as "Windows 8.0" for Windows machines -- this is due to a bug in the JUCE library function for querying the operating system type.

Given that there wasn't anything obvious that caused the crash, my only recommendation is to update to the latest GUI version (0.5.4) and run some test recordings. If you can record for the same time intervals that triggered the crash initially, that would be ideal. We fixed a few bugs related to the Record Node that may improve stability; a full list of changes can be found here: https://github.com/open-ephys/plugin-GUI/releases/tag/v0.5.4

ntotah, 2021-05-12:
Hi,Thanks. We will try that out and keep you posted on the result.Nelson From: Josh Siegle ***@***.***>Date: Wednesday, 12. May 2021 at 16.23To: open-ephys/GUI ***@***.***>Cc: ntotah ***@***.***>, Author ***@***.***>Subject: Re: [open-ephys/GUI] open ephys crash (#387)Ok, it looks like the OS version is always saved as "Windows 8.0" for Windows machines -- this is due to a bug in the JUCE library function for querying the operating system type.Given that there wasn't anything obvious that caused the crash, my only recommendation is to update to the latest GUI version (0.5.4) and run some test recordings. If you can record for the same time intervals that triggered the crash initially, that would be ideal. We fixed a few bugs related to the Record Node that may improve stability; a full list of changes can be found here: https://github.com/open-ephys/plugin-GUI/releases/tag/v0.5.4

closed 2022-07-30 by jsiegle