open-ephys/GUI #388: Possible to automatically start recording to new file after time limit

lstrgar, 2021-05-12:
Hi. Is it possible to set a time or file size limit on a recording being written to disk and once that limit has been reached automatically start writing to a new file? 

Thanks in advance!

jsiegle, 2021-05-12:
There are two ways to do this:

1. Insert a [Network Events](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Network-Events.html) plugin into the signal chain, and use a separate process to start/stop recording at a specified interval. You can use the `control` package in [open-ephys-python-tools](https://github.com/open-ephys/open-ephys-python-tools/tree/main/open_ephys/control) or [open-ephys-matlab-tools](https://github.com/open-ephys/open-ephys-matlab-tools/tree/master/open_ephys/control) for this.

2. Insert a [Record Control](https://open-ephys.github.io/gui-docs/User-Manual/Plugins/Record-Control.html) plugin after a data source, and set it to start recording whenever a particular digital input line goes high.

lstrgar, 2021-05-12:
Thank you, Josh.

lstrgar, 2021-05-12:
Hey Josh. Trying (1) as described. The GUI is now crashing on open after adding the Network Events module into the signal chain. I've tried uninstall / reinstalling. Are there any specific config files I can manually delete? 

Any other advice wrt/ the crash?

lstrgar, 2021-05-12:
FYI I'm just using the .dmg file to install on my mac mini.

jsiegle, 2021-05-12:
What does your signal chain look like?

I can replicate the crash if there's an LFP Viewer after the Network Events module, but not if Network Events is the only plugin in its branch of the signal chain (which is all that's needed to control the GUI remotely).

lstrgar, 2021-05-12:
At this point I can't open the program and do anything without it crashing. Pressing ok or cancel doesn't get me anywhere except to a crash. Screenshot below. Also this is the output system dialogue provides after the crash.

[image]

[image]

jsiegle, 2021-05-12:
I see. You can restore the GUI to a fresh state by deleting `lastConfig.xml` and `recoveryConfig.xml` from /Users/<username>/Library/Application Support/open-ephys

Is it crashing after you press "cancel"?

lstrgar, 2021-05-12:
Yes crashing after cancel.

jsiegle, 2021-05-12:
Weird, I'm not seeing the same thing on my Mac.

In any case, it should work if you have an acquisition board connected. And you can test starting/stoping recording with the File Reader – there are some example files included in the .dmg if you view it in "list mode" (⌘2)

lstrgar, 2021-05-12:
Okay got it. I'm curious to try the Network Events module one more time. Do you have an example of how this should be setup in the signal chain? 

By the way, GUI was successfully reset after deleting those files, thanks for that.

jsiegle, 2021-05-12:
The way you had it is correct...therefore I think the crash was due to the Rhythm FPGA plugin not being able to find a connected acquisition board.

lstrgar, 2021-05-13:
Thanks for your help. I think the GUI just got stuck in a weird state.. might have been my fault.

closed 2021-05-13 by lstrgar