open-ephys/GUI #390: TTL output has random 0 and 5 V fluctuations

mtantirigama, 2021-06-10:
Hi, 
I am recording the TTL output (TTL 1 or 2) via an I/O Board on to a different software, but when I start acquisition on OpenEphys the amplitude of the TTL signal fluctuates between 0 and 5 V randomly. I've attached a picture showing what it looks like on an oscilloscope during acquisition. When I stop acquisition the TTL signal sits idle at either 0 or 5 V depending on when i stop. I am not intentionally sending any TTL output signal from the acquisition board and I'm using a standard system for recordings (64ch intan headstage and Intan Rhythm FPGA Interface for receiving data). What could be the problem ? 

Any help would be appreciated. 
 
[image]

jsiegle, 2021-06-10:
@aacuevas, any idea what could be causing this?

jonnew, 2021-06-10:
Are you sure you have a hdmi connector attached to the digital output and
not the input, which would be hi-Z and therefore might randomly fluctuate?

mtantirigama, 2021-06-10:
yes, its connected to the digital output (second from the right). Also on the intan FPGA interface if I disable the DAC TTL button the signal goes away.

jsiegle, 2021-06-10:
Ah ok, that explains it. DAC TTL translates threshold crossings on one of the headstage channels into digital outputs. If you're not using this feature, you can just disable it.

mtantirigama, 2021-06-10:
oh cool, i'll try it out. Thank you very much for the help !

closed 2021-06-25 by mtantirigama