open-ephys/GUI #391: GUI crash

ChangQing-SH, 2021-07-22:
The first time after I installed the GUI, it works well, but now I don't know why it crashed when I click the .exe, and no errors show in the terminal, and GUI doesn't show in the desktop window,  the only difference is that I install vs2019 today, and try to compile the tracking_plugin. 

OS: win10
GUI: ver 0.5.4

Btw, How to compile the plugin? I follow the docs and also copy the headers into the source folder, but it still displays that can't find the header file, like PlugInfo.h ...

jsiegle, 2021-07-23:
Hi! Which version of the Tracking Plugin did you try to compile? We have a [forked version](https://github.com/open-ephys-plugins/tracking-plugin) that's compatible with the latest version of the GUI. Make sure you place this directory within another directory (e.g. OEPlugins) at the same level as the plugin-GUI folder.

Also, here's a link to the pre-compiled version of the Tracking Plugin: https://www.dropbox.com/s/eon8g6cx4slu57o/tracking-plugin-win64-apiv7.zip?dl=0

Can you try copying that into the "plugins" folder and checking if that works?

ChangQing-SH, 2021-07-23:
hi Josh，
thanks, I figure out the crash issue by delete some files in programdata folder, I use the version from open ephys plugin github, and.also.create a folder at the.same level with gui, the gui can be compiled sucessfully, but the tracking plugin doesn't, anyway i'll try your precompiled version. thanks

btw, any dev plan for a camera plugin for the GUI? so currently it is.better to use Bonsai?   and for.most people do.the recording, it is good.to.make a general.configuration for that

----------

该邮件从移动设备发送

-----------------------------------

jsiegle, 2021-07-23:
We are considering making an official release of [this plugin](https://github.com/rhayman/PosTracker), which supports camera acquisition via the OpenCV library.

However, Bonsai will always be much more powerful and versatile when in comes to video tracking. So in general we recommend that people use it in parallel with the Open Ephys GUI, and connect the two either via the Tracking Plugin, or by triggering TTL outputs from Bonsai that can be fed into the Open Ephys acquisition board.

jsiegle, 2022-07-30:
The [Tracking Plugin](https://github.com/open-ephys-plugins/tracking-plugin) is now available via the plugin installer in GUI version 0.5.X.

We haven't updated it to work with the latest version (0.6.X) yet, but we plan on doing that soon.

closed 2022-07-30 by jsiegle