open-ephys/GUI #392: openephys GUI doesn't open (white screen no error messages)

yves-weissenberger, 2021-08-18:
last line in the console is Num engines:2 --> reinstalling doesn't help 

has anyone encountered this issue?

jsiegle, 2021-08-18:
If you delete `lastConfig.xml` and `recoveryConfig.xml` from `C:\ProgramData\Open Ephys`, does it fix the problem? Those are not overwritten when reinstalling.

veronikasamborska1994, 2021-08-18:
I deleted the Open Ephys folder altogether and that hasn't helped :(

jsiegle, 2021-08-18:
Were you able to run the GUI successfully at any point on this computer?

veronikasamborska1994, 2021-08-18:
Yes, this is the really strange thing. It was working just an hour ago while I was recording from the same setup. It's very hard to understand what happened as there don't seem to be any errors.

jsiegle, 2021-08-18:
Are you installing from a .zip file or via the installer executable?

veronikasamborska1994, 2021-08-19:
so this was extremely stupid but I guess it might be useful for people who don't use windows machines for anything other than recording data like me. I still don't entirely understand why but despite the GUI being opened the actual window was hidden so it was just a matter of clicking nervously (though I am sure there is a specific shortcut) on things for it to pop back up. thanks a lot for your help though! :)

closed 2022-07-30 by jsiegle

lachioma, 2022-11-07:
It happened something similar to me: the GUI would not show up, only the console, and I could see just a white screen of the GUI when previewing it at the taskbar (hovering the mouse on the open ephys app in the taskbar), nothing else. In my case, I had a second monitor connected but not in use. I solved by disabling that screen in the Windows screen settings.