open-ephys/GUI #396: Application not able to keep up with the hardware acquisition

hamdanspam, 2022-02-03:
I am using the gui to acquire neuropixel data with the NI-DAQmx plugin. I am testing out the plugin with a simple pipeline shown below:
[image]

When I press run, the LFP viewer tab lags significantly and then stops altogether: over the course of ~4 s, only ~0.25 s of the display is updated. I get the following error from the debugging terminal.

```
DAQmx Error: The application is not able to keep up with the hardware acquisition.
Increasing the buffer size, reading the data more frequently, or specifying a fixed number of samples to read instead of reading all available samples might correct the problem.
Property: DAQmx_Read_RelativeTo
Corresponding Value: DAQmx_Val_CurrReadPos
Property: DAQmx_Read_Offset
Corresponding Value: 0

Task Name: AITask_USB20229962

Status Code: -200279
Removing audio callback.
Disabling processors...
Disabling Audio Node
Disabling Message Center
Disabling NI-DAQmx
Source node received disable signal
SourceNode returning true.
Disabling LFP Viewer
```

Tech specs:
Windows 10
Intel (R) Core(TM) i7-6700 CPU @ 3.40 GHz
NIDAQmx board USB-6221
v0.5.5.1

I just added two more sticks of RAM, bringing the total memory up to ~32 GB. All drives (C:, D:, etc.) are SSDs with ample room, and the issue persists.

I assume I have to change the buffer size or something like that but the only source for this error (-200279) I could find was LabView specific: https://knowledge.ni.com/KnowledgeArticleDetails?id=kA00Z0000019KTeSAM&l=en-US. 

Is there a way to change the buffer size? I know this NI board is not one of the few that have been verified to work with the gui- is it possible that it sends too much data too fast and I just need to get a different board?

medengineer, 2022-02-04:
Hi,

The plugin currently uses a smaller buffer size for USB devices. Can you try a custom version from [here]( https://www.dropbox.com/s/vihn228ki9kd87e/nidaq-plugin-NI6221-BNC-test.zip?dl=0) where I’ve removed this restriction? Just replace the nidaq-plugin.dll in the plugins directory where open-ephys.exe lives.

hamdanspam, 2022-02-04:
Hi @medengineer, thanks for your response! 

It looks the nidaq-plugin.dll does not live in the same directory as open-ephys.exe ( C:\Program Files\Open Ephys\plugins\), rather it lives in the ProgramData directory (C:\ProgramData\Open Ephys\plugins\).

I assume I should replace the one in the ProgramData directory, but I wanted to confirm with you.

hamdanspam, 2022-02-04:
After replacing the .dll in the ProgramData directory, the LFP viewer briefly updated more "efficiently" (The screen changed faster but still stopped.) The error I got was the same as before, but did not report shutting down the process until I did so manually:

```
DAQmx Error: The application is not able to keep up with the hardware acquisition.
Increasing the buffer size, reading the data more frequently, or specifying a fixed number of samples to read instead of reading all available samples might correct the problem.
Property: DAQmx_Read_RelativeTo
Corresponding Value: DAQmx_Val_CurrReadPos
Property: DAQmx_Read_Offset
Corresponding Value: 0

Task Name: AITask_USB20229962

Status Code: -200279
```

I'm considering shelling out for the BNC-2110 (https://www.ni.com/en-us/support/model.bnc-2110.html)

hamdanspam, 2022-02-04:
One last comment: the task manager reports an increase in CPU activity even after the error is reported and goes down after I stop it running. I guess the DAQ is still sending information and the computer is processing it, but it's not showing up on my LFP viewer.

medengineer, 2022-02-04:
Thanks for trying that out. Does lowering the sample rate in the NIDAQ plugin have any effect? 

The BNC-2110 is optimal, but does also require the PXI chassis (e.g. 1071) and a compatible DAQ card (e.g. 6133). 

If you’re open to trying out a few more fixes, I’m happy to continue debugging over email: [pavel@open-ephys.org](mailto:pavel@open-ephys.org).

hamdanspam, 2022-02-04:
Hi @medengineer, thanks for the idea. It only works if I lower it to 1000 Hz (even 2000 Hz errors out), which may be enough for my purposes. It also only updates the display once every second. 

I should not be surprised, from [this](https://www.ni.com/en-us/support/model.usb-6221.html) website, the total output is only 833 kS/s across all channels, so of course it can't support anything more than 1000 Hz/channel.

closed 2022-02-07 by hamdanspam