open-ephys/GUI #400: GUI crashes when recording a second experiment with Neuropixels probe

yaxigeigei, 2022-04-26:
Hi, I am a new user of the Open Ephys GUI. I installed v0.5.5.3 fresh to a PC this morning. This PC has been recording NP data with SpikeGLX for over a year, and everything works fine. With Open Ephys, I was able to start record using a Neuropixels NHP probe in air. However, if I stop the ongoing recording and restart (which creates a new "experiment"), the GUI crashes. I recorded a [short video](https://www.dropbox.com/s/fjgrte2vwatiqxp/Open%20Ephys%20GUI%202022-04-26%2015-08-26.mp4?dl=0) that demonstrates this process. The problem is very reproducible. 
Btw, after crash, the STAT indicator turns violet.
[image]
Looking forward to your diagnosis!

jsiegle, 2022-04-27:
Hi there! Thanks for posting this issue.

Can you try again with a preview version of the next release, available [here](https://openephysgui.jfrog.io/artifactory/Test-Installer/windows/Install-Open-Ephys-GUI-v0.6.0-rc3-beta.exe)? There are many improvements in this version, especially related to Neuropixels. We still need to finish up the documentation prior to the official release, but otherwise it's pretty much finalized. Please let me know if that fixes the problem.

yaxigeigei, 2022-04-28:
> 

Hey Josh, I tested out v0.6.0rc. The buffer overflow and crash still happens when recording a new "experiment". However, stop and restart using the record button (which creates a new "recording" instead of "experiment") seems fine. I demonstrated the full process from beginning to crash in this [video](https://www.dropbox.com/s/2bxczq3n5u7sxtr/Open%20Ephys%202022-04-28%2010-52-40.mp4?dl=0). 

Another thing that stops me from using the new version is the lack of plug-ins. Specifically, I want to use the Event Triggered Average to assess stimulus related activities online.

There are other minor issues:
When we open the GUI for the first time and select the Neuropixels preset, if the PXI box is not connected, the GUI will stuck at "Scanning for devices" and freeze. Restarting the GUI with PXI box connected won't help. One has to uninstall and reinstall the GUI, and opens it with everything connected to make it work.
GUI freeze when clicking the play button for the first time. Default Neuropixels configuration. Messages in console all normal.

jsiegle, 2022-04-28:
Thanks for sending the video – we will repeat the same sequence of actions and see if we can replicate the crash. When it crashes, does the new "experiment" directory get created, or does it crash before then? If the experiment directory does get created, is there a `settings.xml` file associated with it? If so, please send that to us so we can take a look.

We have moved the Event Triggered Average plugin to a separate repository, and renamed it "[Online PSTH](https://github.com/open-ephys-plugins/online-psth)". We haven't yet upgraded it to work with 0.6.0, but we're planning to wrap that up soon.

If the PXI box is not connected, the expected behavior is for the Neuropixels plugin to launch in simulation mode. Do you know what version of this plugin you have installed? (it should be listed in the settings.xml file). If it's not the latest version, then you will have to uninstall/reinstall through the Plugin Installer to update it.

yaxigeigei, 2022-04-28:
Yes, the crash can be avoided if I always click play before record. The video is [here](https://www.dropbox.com/s/xgczdxiv29p5hu8/2022-04-28%2014-56-04%20crash%20avoided.mp4?dl=0). One thing I noticed was that in the Record Note box the sync indicator for NIDAQ sometimes wouldn't turn green for the first time "playing" (around 0:20s in the video), unless I click on it. Sometimes none of the three indicators turn green, though I didn't catch that in screen capture.

yaxigeigei, 2022-05-02:
Thanks @anjaldoshi @jsiegle ! I tested the new version and the problem is now resolved.