open-ephys/GUI #402: Cannot record from more than 128 channels at a time on OE format

Claudiobasile, 2022-05-13:
I'm part of a research group that started open ephys acquisition system many years ago, when we bought the first acquisition board, using the GUI on windows and recording from 96 channels at first, and then adding 3 more modules of 32 channels each for a total of 192 recording channels. We performed these recording sessions, with a high numbers of simultaneously recording channels, with an old version of the GUI, 0.4.5, which worked perfectly. 

Now, for many reasons, we decided to switch to the new version and use it on Linux Mint. A few day ago we tried connecting the 32 channels modules to the acquisition board to see if the new GUI could visualize and record the activity (we tried that "offline", thus connecting the headstage cables to the acquisition board but not to the EIB, just to see if we could visualize a "noise" signal). We noticed that the new GUI has a separate plug in for recording, and by using that we could not visualize and/or record the "activity" acquired if there where more than 129 channels connected. 

Thus, if we connected only 4 headstages, each associated to 32 channels, to reach a total of 128 channels, we could visualize, record and save the results in the Open Ephys format, but starting from the fifth it gave us problems not allowing any of that. This happened only if we used the open ephys format, and it always switched to the binary format.  

The problem arose even if, with all the modules connected to the acquisition system, we selected from the recording plugin only a few channels. We also tried to install and use the older version that I mentioned, also on linux, and this worked fine with all the modules connected simultaneously.  So we thought it could be a software bug of some sort associated to the newer version, or some kind of error we do when setting the Record Note, maybe duplicating the channels recorded. I will attach an image showing the message appearing when we try to record from 128+ channels.
 
[image]
``

medengineer, 2022-05-13:
Hello,

This is indeed a software bug as it seems the RecordNode is overestimating the number of input channels and throwing the warning.

If you send an email to [support@open-ephys.org](mailto:support@open-ephys.org) we can send over a custom v0.5.5.4 binary that removes this check and will allow you to record all 128+ channels in OpenEphys format.

Claudiobasile, 2022-05-13:
Thank you very much for your response, I wrote just now. Hope to receive it soon in order to test it next week!

medengineer, 2022-05-13:
Updated Ubuntu binary lives [here](https://www.dropbox.com/s/f3ku9mcp92qnx8q/open-ephys-v0.5.5.4-remove-warning.zip?dl=0) in case anyone else runs into this.

Claudiobasile, 2022-05-23:
I'm sorry, but we are trying to install the GUI using this binary file without success. Can you please tell us how to use it in Ubuntu?

medengineer, 2022-05-25:
Here's a [link](https://www.dropbox.com/s/uami35rvt3n4zlq/open-ephys-gui-0.5.5.4-no-warning.deb?dl=0) to the Ubuntu installer. Running this installer will overwrite your current installation automatically.

jsiegle, 2022-07-30:
The Open Ephys format can now record up to ~8000 channels simultaneously. If more channels are needed, you can use the [Binary Format](https://open-ephys.github.io/gui-docs/User-Manual/Recording-data/Binary-format.html).

closed 2022-07-30 by jsiegle