open-ephys/GUI #404: Two+ headstages leads to AUX issues

neuronikki, 2023-03-09:
- Using v0.6.3 OE GUI
- Windows 10
- two+ 16ch Intan Headstages with Accelerometers

Only when we have 2+ Intan headstages plugged in (seperate tethers for recording from 2 subjects), the first headstage (for example plugged into bank A) has corrupt AUX outputs (see screenshot) while B is fine.

A 3rd leads to similar - banks A+B have corrupt AUX but C is fine. 

Has anyone experienced this? 

[image]

jsiegle, 2023-03-09:
This issue has been fixed in the latest version of the Rhythm Plugins – can you try upgrading via the Plugin Installer and see if that solves the problem?

neuronikki, 2023-03-09:
The issue was resolved by running OE GUI as administrator, going to the Plugin Installer and upgrading the Rhythm Plugin, which had an option for Rhythm Plugin Upgrade. Thank you so much for the QUICK reply!

closed 2023-03-09 by neuronikki